ddiakopoulos/hiduino #17: Device not recognized when using Ardiuno Nano (clone)

DeDuckProject, 2015-01-18:
After banging my head against the wall for the past couple of days I've decided to ask this and hope someone has a solution (if there is one at all).

So I want to use an Arduino Nano as my midi device.
I have a few of these so I also use it as an ISP programmer. They are all clones with CH340 instead of the FTDI one (should this affect anything?).

I've flashed HIDUINO_MIDI.hex using avrdude using the following command:
avrdude -p ATMEGA328P -F -P COM5 -c avrisp -b 19200 -U flash:w:HIDUINO_MIDI.hex \ -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

I get a:
verification error, first mismatch at byte 0x000
0xf4 != 0x04
verification error: content mismatch

I don't know if that's such a tragedy but I've tried continuing nevertheless;
I've uploaded, using my arduino programmer, one basic MIDI project - but the device is not recognized, nor sending any midi signals (although it does flash rx when it should).

I've also tried flashing 'usbserial_uno_16u2.hex' but still the same.

One last thing, I've seen somewhere that arduino nano might have a problem with midi over usb because of how its built. If that's any true, which other smaller version of arduino can I use? I have to fit  it into a very small package so Uno is out of the question...

Thanks,
Iftach

ddiakopoulos, 2015-01-18:
Hi there. You are out of luck on the nano. The only supported boards are the Uno, Mega, and Due. The readme states that HIDUINO won't work with FTDI based models (or clones) because the firmware is designed for reprogrammable AVR chips that are used as the USB controller.

One thing I've recommended but not tried myself is to use this board: http://arduino.cc/en/Main/USBSerial

You could try programming that to HIDUINO and connecting the TX/RX pins to your main microcontroller.

DeDuckProject, 2015-01-18:
Thanks for the answer brother - not the best news for me but at least the head-banging can stop now ;)

When talking about the USBserial, do you mean flashing the USB2serial itself and having it communicate with the main microcontroller so that every tx/rx signal is passed forward? That would require some tweaking to the HIDuino code right?

ddiakopoulos, 2015-01-18:
If you look at the architecture of HIDUINO, that's exactly what happens between the Atmega 8u2/16u2 and the "main" Arduino chip that is flashed by the Arduino IDE. The USB2Serial would be a stand-in for boards that don't have the second AVR chip.

closed 2015-04-13 by DeDuckProject

FaSasoftware, 2016-01-03:
Dear ddiakopoulos,
I have another serious problem with your software hiduino. First i'm using Arduino Mega 2560 and if i program it  to send the Midi -Out..Works ok...But i have problems, don't work nothing , when i try to use it as midi in...I don't be able to receive midi in... For example if i program to push a button and send via usb Midi out to reason software or logic studio works fine....But if i play a music from reason to receive midi notes and i program the example led blink don't receive the hiduino midi notes..don't work...In fact i would like to light on my custom leds in my project application with the midi in signals....Can you help me please??? What's wrong with me?? Do you have some simple Midi In example working??Please??Thanks A lot Lestroso.