ddiakopoulos/hiduino #18: Arduino UNO with Mega16u2, not recognized as Midi device! Lufa sources?

tchilaviek, 2015-02-15:
Hello everyone!

I started to learn how to turn my Arduino into a native Midi device with the Hiduino project,

I'm using an Arduino Uno Rev 3 with a 16u2 and a 328p microchip, and an AVR ISP mk2 programmer
after a few tries, I managed to flash my Mega16u2 with avrdude on Windows XP with that command:

avrdude -p at90usb82 -F -P usb -c avrispmkii -U flash:w:arduino_midi.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

it worked grate with no error, but... it's not recognised as a Midi Controller when I plug it with Ableton Live or other software!

I tried it on Windows, Mac, and Linux and the result is the same: nothing happened and the device is not recognised

So I started to explore the MOCO/Lufa and LUFA project

with the mocolufa 'MIDI.hex', and the 'dualMOCO.hex' files, I tried that command:

avrdude -p at90usb82 -F -P usb -c avrispmkii -U flash:w:MIDI.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

and

avrdude -p at90usb82 -F -P usb -c avrispmkii -U flash:w:dualMOCO.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

And ... it worked successfully!

My Arduino is now recognised as a Midi Device, on every platform I tried! even the dual mode with dualMOCO is OK!

my project was to learn also how to compile my own .hex file,
I managed to do that with the MocoLufa Source Code, using WinAVR and the 'make all' command

I noticed that the Lufa source is: LUFA-101122 (for simpleMocoLufa) and LUFA-100807  (for dualMOCO)

Hiduino is using a LUFA-140928 source (latest one!?)

So my question is:

why the .hex files made from a LUFA-140928 source code doesn't fit with my Arduino Uno and Atmel16u2 , while an old LUFA-101122 or LUFA-100807 source based code will work very well on the same architecture with the same tools etc etc??

thank you for your answers!

A.S

Links: 

Hiduino:
https://github.com/ddiakopoulos/hiduino

Moco Lufa simple midi mode:
http://morecatlab.akiba.coocan.jp/morecat_lab/MocoLUFA.html

Moco Lufa dualmode:
http://morecatlab.akiba.coocan.jp/lab/index.php/aruino/midi-firmware-for-arduino-uno-moco/?lang=en

Lufa Library with all the old release include LUFA-101122 and LUFA-100807 (useful for compiling respectively mocolufa and dualmoco sources) :
http://www.fourwalledcubicle.com/LUFA.php

ddiakopoulos, 2015-02-16:
I'm really not sure, and your question is too vague to think of any possible reasons? Is the old HIDUINO noted in the 1.0 tag (and associated LUFA version) working for you? If so, use that.

tchilaviek, 2015-02-16:
Thank you ddiakopoulos for your answer

something strange happened, after you told me to try Hiduino 1.0, I downloaded it, and I flashed the 16u with arduino_midi.hex ,
and it didn't work, it was not recognised on windows xp or windows 7

I tried on Mac OS, and I had a big surprise because it worked very well!!!

then I plugged my arduino back on Windows and it worked!! it was recognised directly,

so I tried to flash arduino with Hiduino 2.0

tried on Mac, it was OK!

and on Windows too!!!

So everything is working now with Hiduino 1.0 and Hiduino 2.0

I do not understand exactly what happened, but I think there is something with the USB device numbers or addresses when plugging and unplugging a device, maybe the system has to refresh or something like that, maybe I have tried to many flashing and I had to reboot my computer 
I don't know!

Thank you for everything!

ddiakopoulos, 2015-02-16:
Yeah, OSX does have a tendency to cache USB devices and their driver profiles aggressively. Glad things worked out for you!

closed 2015-02-16 by ddiakopoulos