ddiakopoulos/hiduino #20: SysEx support

diegoherranz, 2015-03-13:
I think there was a development branch or folder including SysEx support. I can't find it anymore. Has it been merged to main branch? Which is the SysEx support status for v2.0? Any plans?

Thanks!

ddiakopoulos, 2015-03-13:
Yep, there was a contributed firmware that had SysEx support. It's still available in the 1.0 tag. I never really had a use case for SysEx in the mainline. Maybe for v.2.1? Are you looking for full bidirectional, or is host => arduino support enough?

diegoherranz, 2015-03-18:
Ah, now I found it. I need bidirectional but host->arduino is higher
priority.

Thanks!

sanotronics, 2016-09-30:
Where is this branch you found?

Can you post it here please?

mungewell, 2018-06-29:
I have a use case for SysEx messages, sent from Arduino to USB host.

Midi Control Channel (MCC) use SysEx to start/stop/etc a midi sequencer. Some have used these to control an Akai MPC live, which unfortunately does not have hard inputs for start/stop pedals, it will however receive Midi from a class compliant device.

I have started coding this here:
https://github.com/tttapa/MIDI_controller/issues/51
https://github.com/mungewell/MIDI_controller/tree/20180627_DigitalSE

So having SysEx would be quite helpful :-)