ddiakopoulos/hiduino #21: Using Multiple Hiduino Device (Compiled with Name Change)

ibanman555, 2015-03-26:
I have 4 Hiduino programmed Arduino's as midi controller converters.  I have used the compiling descriptions (from Issue #15) and I am able to get each Hiduino to show up as it's own device.  But since I am using more than 1 Arduino, I would like to know what needs to be changed in Descriptors.c and other places to have these be completely independent devices. Because, my program seems to keep cycling through each Arduino looking for "something". There doesn't seem to be an issue when I connect only 2 Hiduino's.

Here is what I changed in Descriptors.c:

Line 56 - .VendorID               = 0x03EB,
Each Arduino has the above code, unchanged
Line 57 - .ProductID              = 0x2049,
I have given each Arduino a different number above.

Line 262 - 267 - USB_Descriptor_String_t PROGMEM ManufacturerString =
{
	.Header                 = {.Size = USB_STRING_LEN(6), .Type = DTYPE_String},

	.UnicodeString          = L"ES8100"
};

I've changed the above to be the same for all 4 Arduino's

Line 273 - 278 -USB_Descriptor_String_t PROGMEM ProductString =
{
	.Header                 = {.Size = USB_STRING_LEN(12), .Type = DTYPE_String},

	.UnicodeString          = L"Controller 4"
};

I've changed the above to have 4 different controller numbers

ddiakopoulos, 2015-03-26:
I don't think this has anything to do with hiduino src itself. I've run 8+ devices on a single computer in the past. If I had any guess, it might be some usb device caching in whatever operating system you are using. I was on OSX.

ddiakopoulos, 2015-03-26:
I also vaguely remember lufa's USE_INTERNAL_SERIAL also being helpful if set manually.

ibanman555, 2015-03-26:
Thanks for the insight.  Can you help by elaborating on how to set USE_INTERNAL_SERIAL manually?

abcminiuser, 2015-04-25:
Sorry Scott - been working through my email backlog (moved a few weeks ago, didn't have internet for a while and now I have a million emails to go through).

That behavior sounds like the boards are disconnecting rapidly, or at least confusing the heck out of the host driver. The code you sent me shows you are using `USE_INTERNAL_SERIAL` as your device descriptor's serial number, so LUFA will substitute the unique device serial into the descriptor when it is requested by the host. Consequently, I don't think it's related to that.

What shows in your Device Manager when all the boards settle down?