ddiakopoulos/hiduino #27: Arduino UNO is not recognised as USB-MIDI device despite successful flashing

dturner, 2015-08-13:
I am able to flash my UNO with `arduino_midi.hex` using the following command: 

`avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -c arduinoisp -p m328p  -v -Uflash:w:arduino_midi.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U lock:w:0x0F:m`

However, when I plug in the Arduino UNO via USB it is not recognised as a MIDI device on OSX. Here's a screenshot from System Information which still shows the name of the device as "Communication Device" rather than "HIDUINO": 

[image]

The only strange thing I did spot is that the final output of `avrdude` is this:

    avrdude: safemode: lfuse reads as FF
    avrdude: safemode: hfuse reads as D9
    avrdude: safemode: efuse reads as 4
    avrdude: safemode: Fuses OK (H:04, E:D9, L:FF)`

The last line reports "H:04 and E:D9" whereas the 2 previous lines report hfuse=D9 and efuse=4 (the other way round).

Is there a way to verify that the flashing operation completed successfully?

ddiakopoulos, 2015-08-13:
All of those things you indicate point to a situation where you have flashed the 328 chip on the R3 itself, not the USB controller. There are 2 sets of 6 pin headers. 

The connector should be positioned as follows:

http://cdn.instructables.com/F2E/MQDM/GS8OSZXF/F2EMQDMGS8OSZXF.LARGE.jpg

From the Arduino IDE, however, the function "Upload Using Programmer" is really helpful to flash sketches onto the 328 while the other USB chip has been flashed by Arduino, since HIDUINO would have replaced the serial bootloading function of Arduino.

dturner, 2015-08-13:
Ah, now it all makes sense! I had attached the Arduino ISP to the wrong headers. Once I changed it everything worked correctly. Here's the avrdude command string I used: 

    avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -c arduinoisp -p m16u2 -U flash:w:arduino_midi.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

And for anyone else using the Arduino ISP here's a photo of it correctly installed for flashing the USB controller: 

[image]

Many thanks to you ddiakopoulos for the quick and helpful response :)

dturner, 2015-08-13:
And for good measure here's the System Information window once the arduino USB firmware has been flashed: 

[image]

closed 2015-08-13 by dturner