ddiakopoulos/hiduino #28: 32u4 leonardo

888noxxi, 2015-09-18:
Hi, I'm having trouble getting hiduino to work on my 32u4, i need it to basically act as a usb midi converter, I thought I could just upload the hiduino firmware to it and have it work but I cant get the PC to even respond to it, flashing the leonardo firmware back works fine, and I've mabaged to get it to work no problem on an 8u2 and a 16u2 but not the 32u4. 

Is there anything I'm doing wrong? (I'm using an ISP programmer)

ddiakopoulos, 2015-09-18:
Unfortunately the Leonardo is a single chip board. Even though you can flash the hiduino firmware, where will your Arduino sketch run? The Uno and Mega boards are unique since they use a separate 8u2 or 16u2 for USB, while the actual user-uploaded Arduino sketch runs on the "bigger" chip.

888noxxi, 2015-09-18:
Hi again, thanks for getting back to me.

All I need it to do at the moment is to receive midi on its rx pin from another arduinos tx pin, and forward that to the usb port. 

Essentially, I am trying to us the 32u4 leonardo pro micro like it is a 16u2 or 8u2, the reason is that I am trying to make a midi clock with 2 usb ports for 2 laptops, I have an uno with the sketch on it (but its a clone with a ch340) its outputting serial midi on its tx pin, I want the 2 leonardos connected to that pin, only forwarding the serial data to the usb ports.

Thanks again
Allan

ddiakopoulos, 2015-09-20:
Hi Allan. I'm not sure using a Leonardo is the right approach (it's a board I've never supported before -- I think bringing it up with the firmware is more trouble than it is worth). My recommendation is to use two of these boards, one for each of your laptops:

https://www.arduino.cc/en/Main/USBSerial

It is the most simple 16u2 breakout with all the functionality to tx/rx to a common clock running on your Uno clone. Sorry my solution requires extra hardware!

888noxxi, 2015-09-21:
Thanks for your advice,
Your most likely right, I'v got a couple of these leonardo pros kicking about at the moment, SO i'm going to experiment with them a bit and maybe learn how to create something with lufa. for the moment though i'll probably use the 16u2 breakout, just to get something up and running in the mean time. 
My next thing with these leonardos is to try install the cloned teensy bootloader and write a sketch to forward the serial that way. 

Thanks for your help.
Allan

888noxxi, 2015-09-21:
I just wanted to add a solution that I came across whilst looking around for lufa information.

I'm using the 32u4 Arduino Leonardo Pro Micro Clone

I found a hex file for the 32u4 here:
http://morecatlab.akiba.coocan.jp/morecat_lab/MocoLUFA.html

Using an ISP Programmer and wired up like this;
http://www.coreforge.com/blog/2014/09/recovering-arduino-pro-micro-atmega32u4/

and using this line in AVRdude;
avrdude -c usbasp -p atmega32u4 -F  -U flash:w:MIDI.hex -U lfuse:w:0xFF:m -U hfuse:w:0xD9:m -U efuse:w:0xF4:m -U lock:w:0x0F:m

Not changing the fuse back, when it asks.

It seems to work, at least it works fine for my needs anyway.

I'm just putting this information here in the hope its useful to someone. Its not a slur on your work, which is superb and without it I could never made my last controller. If posting this here is in any way inappropriate then please let me know.

A big thanks for taking the time to look at my issue and reply to it. Is there any way I can buy you a beer? like can I donate a few bucks or something?

Allan

ddiakopoulos, 2015-09-21:
Cheers. I will add a note in the readme about your findings :)

closed 2015-09-21 by ddiakopoulos