ddiakopoulos/hiduino #31: Mega2560 is not recognised as USB-MIDI device

Carlos10240, 2015-10-03:
Hello,

I compiled the arduino_midi again with my name"carlos-midi' and flashed with a Mega2560 using atmel studio on Win7 64 bit.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

All went smoothly, but still get Mega2560 when plugin.

I hereby attached the process in order:
1.	Plugin avrispmkII
2.	preparing atmel studio
3.	setting fuse
4.	setting lock bit
5.	flash and verify
6.	plugin USB
7.	Arduino Mega 2560 in Device manager

I would need your help to develop a usable usb-midi device with Arduino, but now I get stuck at the first stage, what should I do?

Thanks a lot,

Carlos

diegoherranz, 2015-10-04:
You have to flash the other Atmel chip on the board, the ATMEGA16U2/8U2 which is the one doing the USB-serial conversion and not the main one. Connect the avrispmkII to the other header connector, the one close to the USB port and try again.

Cheers!

Carlos10240, 2015-10-04:
Oh yes!  I follow your instruction and successfully see carlos_midi when plugin!  Now I know it is the atmega16U2 doing the USB-serial conversion but not the Atmega2560.

However, that make me confuse.  Would you please tell me when I program the arduino with sketches(pde), am I programming the Atmega16U2 or the Atmega2560?

Later, when I am going to develop that "carlos_midi" USB with sketches, should I refresh the atmega16U2 with its original hex everytime?

Carlos

diegoherranz, 2015-10-14:
> However, that make me confuse. Would you please tell me when I program the arduino with sketches(pde), am I programming the Atmega16U2 or the Atmega2560?

The Atmega2560. The Atmega16U2 just performs the USB-serial conversion (or USB-MIDI in this case).

> Later, when I am going to develop that "carlos_midi" USB with sketches, should I refresh the atmega16U2 with its original hex everytime?

To program the Atmega2560 using the standard procedure, you would need to flash the Atmega16U2 with the standard serial-USB HEX every time, which is a pain. Another option is connecting the avrispmkII to the ISP header of the Atmega2560 (the header you used the first time) and flash it directly without bootloader. You will need to flash the booloader back to the Atmega2560 to go back to normal operation. Instructions here: https://www.arduino.cc/en/Hacking/Programmer

This is described too on this project's README: https://github.com/ddiakopoulos/hiduino#quickstart

closed 2016-01-09 by ddiakopoulos