ddiakopoulos/hiduino #32: USB Serial and HIDUINO at the same time

sanotronics, 2015-11-25:
Hello!

Would it be possible?

I would love to, for example, have switch, that the Atmega looks at, at startup, and runs one or the other.

What do you think?

ddiakopoulos, 2015-11-25:
Difficult! 

I did some preliminary work on this 2+ years ago in toggling between USB serial and HIDUINO by checking the state of the reset pin on a 16u2 chip when it powered on (there are only 6 pins directly corrected to the 16u2 for programming... using reset pin is definitely a hack). I seem to remember that this worked? 

The biggest problem was fitting both HIDUINO and the USB serial stack on the 16u2. It is definitely *not* possible on any configurations with an 8u2 since there's just not enough memory for both programs. I also remember something about USB endpoints being an issue.

There's definitely some hacking to be done here, but unfortunately I don't have time to tackle it myself.

sanotronics, 2015-12-05:
Thanks dimitri!

So it worked? you could fit both of them in the 16u2 and switch with the reset pin at power on?

I could take a look at some point and try to make it work if you can give me whatever you have so far :)

ddiakopoulos, 2015-12-18:
I just found out that the author of MocoLufa (https://github.com/kuwatay/mocolufa/blob/master/dualMoco.c) has an implementation of this.

closed 2016-02-05 by ddiakopoulos