ddiakopoulos/hiduino #33: Hiduino recompiler web app

sanotronics, 2015-12-05:
Hello!

@pablofullana has developed an early stage of a tool that I think could help lots of hiduino users.

It is a [web app](https://f8610f89.ngrok.io/) for recompiling and renaming the hiduino firmware, to show up as an HID with any name we want.

[This is the project rep](https://github.com/pablofullana/hiduino_config)

It is in absolute alpha version, and I am hosting it in a Ubuntu desktop and publishing it with ngrok, I will leave it like this for a couple of days so you can try it out.

For now it works with one request at a time, and only in Firefox, can't figure out the reason for this last part yet.

If some of you guys want to help and host it in a server you own and make it more permanent it would be awesome.

Also if you want to contribute just generate a pull and be brave :)

Hope you like it!!

ddiakopoulos, 2015-12-13:
Whoa! This is awesome... nice work! If I had one functional comment, the fields for name and manufacturer should be capped to 11 characters which is the maximum that the USB descriptor can handle.

ddiakopoulos, 2016-01-10:
Closing this out for now. Maybe in the medium-term future I will host an instance of this application myself and put a link a readme. Thanks again for your hard work!

closed 2016-01-10 by ddiakopoulos

sanotronics, 2016-04-02:
Ok, magicly I handled myself to deploy this on AWS, and now we have a permanent link

[http://pimpmyhiduino.yaeltex.com](url)

:)