ddiakopoulos/hiduino #34: 'MIDI' was not declared in this scope - unresolved!

BadMuthaHubbard, 2015-12-09:
Hi!
I see that someone else wrote with this problem a while ago and closed the issue after you suggested a particular MIDI library. But the problem is not that the library is not loaded; the problem is that it is not declared in the scope in which it is used, in the first example. I get this error for both MIDI.begin in setup(), and MIDI.sendNoteOn in loop().
I installed the exact MIDI library you recommend, but apparently it's the way the first example is written. Could you take a look? My C is a little rusty, but I will look it up when I get a chance.
I'm trying with Arduino 1.6.6 on Win 7 (I tried also with Arduino 1.0.5).

Thanks.
Chuckk

ddiakopoulos, 2015-12-18:
Hi, I just updated the examples to use the latest Arduino MIDI library: 55e2c1a1e7fa40e427f2f560e6691a81fa28aef8. If you pull the latest HIDUINO, it should work for you, but note the new library.properties requirement for 1.6.x IDEs that I reference in the updated readme (you'll likely need to add one additional file to the arduino_midi_library folder).

closed 2016-01-09 by ddiakopoulos

BadMuthaHubbard, 2016-01-10:
Thank you for replying and for the information! We took in a foster daughter right after I asked the question and I haven't had 3 free minutes yet to reply.

-Chuckk

Date: Sat, 9 Jan 2016 13:09:05 -0800

Closed #34.