ddiakopoulos/hiduino #35: Hiduino Midi In Don't work with Arduino Mega 2560

FaSasoftware, 2016-01-03:
Dear ddiakopoulos,
I have another serious problem with your software hiduino. First i'm using Arduino Mega 2560 and if i program it  to send the Midi -Out..Works ok...But i have problems, don't work nothing , when i try to use it as midi in...I don't be able to receive midi in... For example if i program to push a button and send via usb Midi out to reason software or logic studio works fine....But if i play a music from reason to receive midi notes and i program the example led blink don't receive the hiduino midi notes..don't work...In fact i would like to light on my custom leds in my project application with the midi in signals....Can you help me please??? What's wrong with me?? Do you have some simple Midi In example working??Please??Thanks A lot Lestroso.

ddiakopoulos, 2016-01-03:
Maybe link to your Arduino sketch in a Github Gist and post it here so I can take a look. A minimal example for receiving MIDI is located here: https://github.com/ddiakopoulos/hiduino/blob/master/example_sketches/midi_input/midi_input.ino

FaSasoftware, 2016-01-03:
Thank a lot Iddiakopoulos for your fast answer....
ok i saw that you have updated your software, but in this moment i can't try that because i'm away from my home....ok i enclose here my program test..that i repeat  works fine in Midi Out, but not in Midi In ...I thank you so much for your  help.Happy new Year 2016. Lestroso.
[Keyboard_3b.zip](https://github.com/ddiakopoulos/hiduino/files/76762/Keyboard_3b.zip)

FaSasoftware, 2016-01-04:
Thank a lot Iddiakopoulos for your fast answer....

Ok i saw that you have updated your software, but in this moment i can't
try that because i'm away from my home....ok i enclose here my program
test..that i

repeat works fine in Midi Out, but not in Midi In .But in truth..your
software don't work for me with any Midi In software..also with the
examples..Do i have perhaps

to update your firmware???..I thank you so much for your help.Happy new
Year 2016. Lestroso.

FaSasoftware, 2016-02-05:
Dear Dimitri Diakopoulos,
Can You Help me please? Do you have a solution for this Midi In problem?? I need as soon as Possible.
But i don't understand , i'm the only one with this problem??
Can you give me a simple skatch.ino Midi-in to try??
Or you mast update your firmware??
Or where Is it My fault?
I thank you a lot in advance.
Best regards,
Lestroso

ddiakopoulos, 2016-02-05:
Yes, MIDI input definitely works (and I haven't touched that part of the code in over a year). Please attach your .ino file here so I can look at your Arduino sketch and check if you are missing anything.

FaSasoftware, 2016-02-06:
Dear ddiakopoulos,
thanks alot for your help...I enclose here my file .ino ....Please confirm
that you have received well my file....Thanks again.
Best regards,
Lestroso

ddiakopoulos, 2016-02-06:
No, you can't attach in email. You need to upload the file to the actual Github tracker on the website.

ddiakopoulos, 2016-02-07:
Ok, you sent me Keyboard3b.ino via email, I am attaching here as a txt file.

What version of the MIDI library are you using? You aren't following the callback convention used by the MIDI library. Please see the `midi_input.ino` example for reference: https://github.com/ddiakopoulos/hiduino/blob/master/example_sketches/midi_input/midi_input.ino

The way you do a MIDI.read() looks wrong. Please experiment with the example file I linked above and start from there. 

[Keyboard3b.txt](https://github.com/ddiakopoulos/hiduino/files/120499/Keyboard3b.txt)

FaSasoftware, 2016-02-07:
Ok... i will do.... i use the MIDI Library 4.2  in Arduino Site.... I'll try to experiment with midi_input.ino Software....Best regards,
Lestroso

closed 2016-03-05 by ddiakopoulos

Gabryxx, 2018-02-06:
Hi, sorry for bad english...
ok..i have arduino uno recognised how midi interface from PC, a Roland D10 keyboard, midi interface with octopuler ecc.. all is work ok (send notes,controllers) but i have problem when i try send notes back to instrument:
RX led on arduino flash when incoming a note from PC but none reach Roland keyboard, but if i try use your example (MIDI_Basic_IO.ino), my Roland play note, so connection and interface is ok.
Also if i put send notes inside mi sketch is played from Roland, but none incoming from PC, also if RX led on arduino receive the message.
What is problem?
Ty for help and library