ddiakopoulos/hiduino #36: Hiduino changing midi message?

markim, 2016-01-09:
Hello,

First of all thank you all the devs (hiduino, LUFA) for your awesome arduino contributions.

Here's my problem:

I used this tutorial to make my midi controller: http://www.instructables.com/id/Arcade-Button-MIDI-Controller/

The software that I have loaded up on the Arduino is here: http://www.instructables.com/files/orig/F31/XWO4/GXL5XQYD/F31XWO4GXL5XQYD.zip

When I enable debug (so I can see human readable serial debug messages) it shows all of my buttons and potentiometers working correctly with push down / up, and pot ranges between 0-127. 

If I use a Serial - > USB converter and a Virtual Midi device, I can open programs like MIDI OX, Ableton, and Mixxx with successful mappings for every part of the controller.

Whenever I flash the Hiduino, though, the arduino is recognized by the system as a Midi compliant USB device, but the MIDI message is being altered. Instead of everything being on channel 1, each button is a signal almost on it's own channel. I am sure the message is getting offset somehow with Hiduino, but I do not know what to do about it. (for example if I hit the 5th button on my controller 3 times it will show in MIDI OX as getting a midi message for channel 4, then the 4th time I hit it will show a midi message on channel 5, then it will repeat the pattern)

If you have any information that might help me fix this I would greatly appreciate it.

Thanks

ddiakopoulos, 2016-01-09:
Hi Markim,

It seems your sketch is composing MIDI messages from bytes directly and there's something lost in translation within the HIDUINO firmware before being re-sent as USB-MIDI. I'd recommend re-rolling the sketch using the Arduino MIDI library (linked in the README) which is known to compose the correct MIDI message format.

markim, 2016-01-10:
Oh, excellent. I had some time this morning and actually got around to fixing it. I should have read directions more carefully!

The same script worked when I did what you said and included the MIDI library, then changed the note on/ note off / cc  from the original script. I deleted the note on/note off / cc methods in the script because they weren't being called any more, then changed them. E.g

noteOn(MIDI_CHANNEL, NOTE + digitalOffset, 0x7F);

turned into 

MIDI.sendNoteOn(NOTE + digitalOffset, 0x7F, MIDI_CHANNEL);

(Had to change the order a little bit <-- that's a pun :P)

Just wanted to say thank you for the help (and to document in case anyone else has this issue). My midi control now a HID-compliant USB device :)

One quick thing: I changed the CC functions to this:

MIDI.sendControlChange(i, analogueInputs[i], MIDI_CHANNEL);

So as it's reading through the i # of analog inputs I have, it's just writing the control change for whatever analog potentiometer I play with. It doesn't really matter which CC number I use, does it? Right now (according to this http://nickfever.com/music/midi-cc-list, and with tests from MIDI OX) my midi CC's are: Bank Select, Modulation, Breath Controller, Undefined, Foot Controller, and Portamento Time. 

Does it really matter which one of these are used if you're making a USB controller? I could imagine wiring up the MIDI port to my project, then plugging it into something else and it being confused, but if it's usb, and all the inputs are programmed before using it, I guess it's up to the user!

Thanks again for the help, have a great year.

ddiakopoulos, 2016-01-10:
Glad it worked out! Thanks for your thorough debugging documentation :+1:

closed 2016-01-10 by ddiakopoulos