ddiakopoulos/hiduino #38: Flip: Address is out of range

simbooper, 2016-02-26:
@ddiakopoulos Hello,
We're currently having trouble with the **usbserial_mega_16U2.hex** for the **Arduino Uno Rev3**.
Flip tells us this error :

> Class com.atmelflipGui.FileMenuHandler
> The Address is out of range

The reason could be that our arduino may not have enough space to put this 20ko file in, because flip says "12 ko" max for our mega16u2. We can't change the maximum size authorized in flip unfortunately (even by changing min and maxin the buffers). We tried different sizes of hex to input, and it appears that our maximum is 16ko... 
Are we doing it right??

PS: we tried some other hex (LUFA mocco), and the arduino is constantly sending midi to the music program we use (FYI Ableton).

lefthand2, 2016-10-17:
I'm having the same "out of range" issue with Uno R3

ddiakopoulos, 2016-10-17:
The usbserial firmware isn't related to hiduino. In fact, these usbserial compiled hex files come directly from the Arduino Github. It is a quite curious that the 16u2 firmware is bigger than the 16kb of memory available on the chip, but I've never had an issue using the ISP based method of flashing.

https://github.com/arduino/Arduino/tree/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbserial

lefthand2, 2016-10-17:
Wow, that's great you got back to me so quickly. Thought I was just lost. 

Is the link above for USB-midi?    I loaded the Arduino-usbserial-atmega16u2-Uno-Rev3.hex file using Atmel Flip, but the Uno now enumerates as Arduino Uno.  I think I'm back to stock firmware on 16u2 chip.  I'm trying to put the usb-midi firmware on the 16u2 chip.

demin-dmitriy, 2016-10-18:
@lefthand2, `usbserial_mega_16U2.hex` file in this (hiduino) repository is actually a `Arduino-COMBINED-dfu-usbserial-atmega16u2-Uno-Rev3.hex` from official `Arduino` repo. Nowhere is written about this fact (or I missed it) so at first I was confused as well (because i thought it was actual hiduino firmware). This file is for some reason too big to fit in FLASH, so Flip (rightly) won't let you load it.

My Arduino Uno R3 initially had `Arduino-usbserial-atmega16u2-Uno-Rev3.hex` (from official Arduino repo) in it's FLASH, so use that to restore 16u2 to it's original firmware.

To actually load HIDUINO use `arduino_midi.hex` file.

lefthand2, 2016-10-19:
Thank-you for explaining that!

Ken

lefthand2, 2016-10-31:
I was able to upload arduino_midi.hex into my Uno 16u2 chip using Flip, but the Uno board doesn't enumerate itself in device manager as Huduino;  it shows up as libusb-win32/ATmega16U2.  If I plug the usb into a new computer, it is looking for ATmega16U2DFU drivers.  I'm a bit confused why device manager doesn't recognize the Uno as a midi device....??

MarcelWierda, 2016-11-20:
@lefthand. I ran into comparable HID functionality errors on the UNO.
May I concluded that buying a Leonardo or Micro  will safe you loads of time. because both have HID 'on board', plus libraries, as it is promised?
GLHF! Marcel