ddiakopoulos/hiduino #42: arduino_midi not recognized in Windows

drder66, 2017-10-24:
I have an arduino mega2560 that I've programmed with hiduino. On the face of it, all works well. On linux kmidimon shows arduino_midi as a device and correctly shows the NoteOn/NoteOff messages I'm sending (this is a Midi output only application).

On Windows, the device shows up in "devices and printers" as an Audio Midi HID device, and Hairless Midi will try to route the output of arduino_midi back to a serial port, showing NoteOn/Note Off mesages if I let it (not that it would be at all useful, but shows that a windows program is able to access the device).

The application I'm trying to use is Hauptwerk. This refuses to recognize the arduino_midi, but shows other USB midi (commercial) devices. I've looked long and hard at device properties, comparing the commercial device and the arduino, and I can't see anything that wouldn't be expected from the two devices not being able to occupy exactly the same spot in the USB hierarchy.

I've been in touch with the Hauptwerk developer, who can't (for fairly obvious reasons) give much practical help debugging the issue. What he did note is that Hauptwerk uses the windows functions _midiInGetNumDevs()_ and _midiInGetDevCapsW_ to enumerate Midi devices present in the system.

Is there any way that you can think of that arduino_midi could be shown as a Midi device in the Windows Control panel and yet not be enumerated by the above two functions ? Other than trying to write a windows app calling those functions, I'm at a loss for ideas. Any ideas helpful. Thanks

drder66, 2017-10-25:
I should add that the Hauptwerk developer also added that his code does nothing other than call the Windows functions to confirm the presence of a Midi device i.e. it doesn't probe the device with SysEx messages (for example)

drder66, 2017-10-25:
I can confirm that midiInGetNumDevs() returns 0 in a system with an arduino_midi being shown as a Midi device in device manager.
The only difference I can see with usbview.exe is that arduino_midi endpoint descriptor address is 2 for output and 1 for input compared to another midi device which has 1 for input and output. 
Any ideas appreicated

drder66, 2017-10-26:
Anyone know how midiInGetNumDevs() figures out whether a USB Audio device is a Midi one to list or not ?

Gabryxx, 2018-02-12:
I have arduino Uno and i use Flip for put MocoLufa Usb/Midi switchable with 4-6 ISP pin, and input midi is recognized in Windows 7 64 from all audio programs.
Problem is that none go back from PC to my midi keyboard (Roland D10)...so notes/sysex etc. start from PC, go to arduino (led RX flash) but not go from arduino to midi interface and instrument.
If i put on sketch the command for play a note or control change, the note reach Roland keyboard, so is no problem of midi interface or connection.